commaai/agnos-builder #11: Filesystem corruption prevention

adeebshihadeh, 2021-11-11:
current rw partitions
- [ ] /persist - we should probably make this ro
  * we put the SSH host keys on there now. these should move to userdata
  * should only need to remount rw in provisioning to write the keys
- [ ] /cache - just handle corruption in agnos by reformatting
- [ ] /dsp - probably doesn't need to be rw?
- [ ] /firmware - also probably doesn't need to be rw?
- [ ] /data - we should try to unmount cleanly
- [ ] /data/media - ditch the last segment and unmount cleanly once we can make sure the rlog ends up on userdata

gregjhogan, 2021-11-11:
The life cycle of the SSH host keys should be the same as the device key pair, so they should be generated at the same time.

adeebshihadeh, 2021-11-24:
I mean this sounds nice, but I don't think it's worth the complexity and risk of remounting. Ideally, we only mount /persist as rw during production while generating the keys. I'd much rather have the guarantees of persist being ro than the host keys never changing.

gregjhogan, 2021-11-24:
We shouldn't be generating any private keys before the user has the device in their possession.

closed 2025-10-04 by adeebshihadeh