commaai/agnos-builder #16: [$5000 Bounty] Use Upstream Weston

VirtuallyChris, 2021-11-15:
We currently use the closed source Thundercomm version of Weston.

Get open source Weston working in AGNOS for a $5000 bounty.
https://github.com/wayland-project/weston

geohot, 2021-11-16:
Has to run openpilot as normal to qualify

robbederks, 2021-11-18:
It actually turns out all the code we use (except for some minor internal changes) is in fact open source, albeit horribly outdated.
For reference, these are the public remotes used to build the weston we use now:
- https://source.codeaurora.org/platform/hardware/qcom/display: `LE.UM.2.3.7`
- https://source.codeaurora.org/platform/external/libdrm: `aosp-new/nougat-release`
- https://source.codeaurora.org/quic/le/wayland/weston: `LE.UM.2.3.7`

kfatehi, 2021-11-22:
Hi. I took a shot at this out of curiosity and failed, but I got some potentially useful output and scripts out of it, having been able to execute a custom built weston 9 on the tici at the end of it. Documented a post-mortem of my attempt here with the relevant error logs: https://github.com/kfatehi/agnos-builder/blob/3a3e0839bbba9c6d1015915e8f999a57e990fdd0/weston/README.md

houqp, 2021-12-05:
@robbederks is the corresponding mesa freedreno gallium driver source also open-sourced?

robbederks, 2021-12-05:
> @robbederks is the corresponding mesa freedreno gallium driver source also open-sourced?

We aren't using freedreno, but rather the proprietary Adreno shared libraries (see the content of the display deb package). Making this part open source is obviously not in the scope of this bounty.

houqp, 2021-12-06:
OK, turns out unlike the open-source mesa gbm, the proprietary version has the gbm driver backend statically linked into the provided `libgbm.so` shared library. @kfatehi your setup linked weston with ubuntu provided gbm lib, which loads driver backend from `lib/dri/*_dri.so` files. This is why you are were getting that `MESA-LOADER: failed to open msm_drm` error.

@robbederks is the proprietary `gbm.h` header file available for share to help with the linking?

robbederks, 2021-12-06:
> OK, turns out unlike the open-source mesa gbm, the proprietary version has the gbm driver backend statically linked into the provided `libgbm.so` shared library. @kfatehi your setup linked weston with ubuntu provided gbm lib, which loads driver backend from `lib/dri/*_dri.so` files. This is why you are were getting that `MESA-LOADER: failed to open msm_drm` error.
> 
> @robbederks is the proprietary `gbm.h` header file available for share to help with the linking?
> 
> UPDATE: looking at the weston fork, it looks like the proprietary gbm is a fork of minigbm, not the one in mesa.

Everything we can open source is either in this repo (agnos-builder) or in the repos I've linked before, and this should be enough afaik.

I do remember looking into this some time ago that there is libgbm and libgralloc (the android version), and I believe they wrote a libgbm-like interface to libgralloc, since that's what adreno uses internally.

robin-reckmann, 2022-03-15:
Rather than forcing upstream Weston into the current AGNOS with the Thundercomm kernel, porting to mainline Kernel is the better options I guess. Then we will get open source graphics drivers and upstream weston for free. I'm currently trying to get everything else also running with mainline. https://github.com/robin-reckmann/comma3_mainline

geohot, 2022-03-15:
The open source graphics drivers don't support OpenCL afaik, which we need to run the model. The other issue with mainline kernel is the lack of well tested camera support.

zorrobyte, 2022-04-27:
Linaro guys say
[image]

martinl, 2023-05-31:
I will slowly start defining the scope and steps required for getting this done by going mainline route.

We have at least one developer interested in developing open source camera drivers for c3 sensor (with some [previous experience](https://emainline.gitlab.io/2022/08/24/camera.html)). But missing OpenCL support is the big question. I was thinking about using something like http://portablecl.org as base for adding hw acceleration using https://gitlab.com/postmarketOS/pmaports/-/merge_requests/4050 
It's currently beoyond my skillset and I would need some help with figuring it out if it is possible. In my opinion and from general perspective it looks promising and worth figuring out as a challenge.

This CDSP guide for 845 might also be relevant https://git.linaro.org/people/srinivas.kandagatla/ImagenetCv.git/tree/README but I'm not sure yet. We'll see.

martinl, 2023-05-31:
TODO
- find a path forward for OpenCL
- get serial console adapter and c3
- get Robin's mainline branch booting on c3
- get gpu working
- get ssc / sensors working
- get cdsp up and running
- start initial development on opensource camera driver for c3 sensor
- get additional c3 or camera for Tooniis for driver development and testing
- develop camera drivers for c3 sensor

robin-reckmann, 2023-05-31:
afaik there's nothing connected to the ssc. Most sensors are accessible via i2c and upstream drivers.

martinl, 2023-05-31:
yes, but at least hexagonrpcd and accessing dsp via qmi and protobuf might be relevant for using adsp and cdsp later

martinl, 2023-05-31:
> afaik there's nothing connected to the ssc. Most sensors are accessible via i2c and upstream drivers.

yes, but at least hexagonrpcd and accessing dsp via qmi and protobuf might be relevant for using adsp and cdsp later

robbederks, 2023-05-31:
> afaik there's nothing connected to the ssc. Most sensors are accessible via i2c and upstream drivers.

correct

okias, 2024-12-15:
I managed to get basic OpenCL support into Mesa3D 24.3 series ( https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/30835/ ).

Generally using hexagon or what qcom has for compute may be better option, but for the start "it's something".

The camera API may be a bigger challenge, cause for now mainline have only D-PHY ‒ we don't have C-PHY support yet (in case it's used by OpenPilot HW).

adeebshihadeh, 2025-08-11:
> Any reason for which this bounty is locked? As mentioned by [@robin-reckmann](https://github.com/robin-reckmann) above, both `downstream kernel`-`Weston fork` and `mainline kernel`-`upstream Weston` form two sensible pairs, and therefore as I'm doing some porting work on the mainline kernel, it makes no sense to even attempt to start with the currently used Weston version. Does this bounty also apply in case of the mainline kernel port?

Unlocked!