commaai/agnos-builder #110: Boot time speedup

adeebshihadeh, 2023-02-18:
AGNOS 6:
* XBL: 3.5s
* ABL: 4.7s
* Kernel: 
* comma.service:

adeebshihadeh, 2023-02-26:
Added a [script](https://github.com/commaai/agnos-builder/blob/master/userspace/usr/comma/tests/analyze-boot-time.py) to print boot times. Here's where we're at as of 1e0f240f38622f752e926a0edb53efe9f23d5833 
```
comma@tici:/data$ ./analyze-boot-time.py
------  ----  -----  ---
PON     1.5   1.5    9%
XBL     2.4   3.9    14%
ABL     3.7   7.6    21%
kernel  3.59  11.19  20%
weston  6.37  17.57  36%
comma   0.01  17.58  0%
onroad  ?     ?      -
------  ----  -----  ---
```

adeebshihadeh, 2024-01-21:
Useful commands
```
systemd-analyze time
systemctl list-dependencies comma.service
```

andiradulescu, 2024-02-03:
Related to locking https://github.com/commaai/openpilot/issues/30894.

My current (early) progress is running `./analyze-boot-time.py` + running the `useful commands` from the previous comment + analyzing `dmesg` with [initcall_debug](https://elinux.org/Initcall_Debug) and generating a `boot.svg` with `scripts/bootgraph.pl`.
```
comma@comma-xxxxxxxx:/usr/comma/tests$ ./analyze-boot-time.py
------  ----  -----  ----
PON     1.5   1.5    10%
XBL     2.4   3.9    16%
ABL     3.7   7.6    25%
kernel  3.54  11.14  24%
weston  6.23  17.37  43%
comma   -2.8  14.57  -19%
onroad  ?     ?      -
------  ----  -----  ----
```
[boot.svg](https://github.com/commaai/agnos-builder/assets/76900/4b4caecf-095b-4016-ada2-f5eb15e3de46)
[image]

closed 2026-10-21 by adeebshihadeh