commaai/agnos-builder #171: Wi-Fi networks with WPA3 are not supported

TheOfficialJosh, 2023-08-20:
Wi-Fi networks using only the WPA3 security protocol are showing the 🚫 symbol next to them, and cannot be joined.

- The Qualcomm 845 does seem to support WPA3 at a hardware level. (Source: [XDA Developers](https://www.xda-developers.com/the-qualcomm-snapdragon-845-will-support-the-new-wpa3-wi-fi-security-standard/))
- Discord user nelsonjchen said the missing Security Flag might be "NM_802_11_AP_SEC_KEY_MGMT_SAE". ([Link to post](https://discord.com/channels/469524606043160576/871838269405556736/1018542403067584605))

See the network "DingoAteYourBaby" in the attached screenshot, which is only using WPA3. The IOT network current connected, has the exact same settings, on the same Access Point, except it is using WPA2.
[image]

adeebshihadeh, 2024-07-26:
@andiradulescu you know if 24.04 fixes this for free?

andiradulescu, 2024-07-26:
As it is, I kind of doubt it. I need to test - switching my router to WPA3 only and let's see what happens when I try to connect from 24.04.

andiradulescu, 2024-07-26:
The WPA3 network appears in the list grayed out with a "forbidden" icon to the left of the right signal icon. I think I need to patch this in https://github.com/commaai/openpilot/pull/27824.

andiradulescu, 2024-07-27:
Patched [wifi_manager.cc](https://github.com/andiradulescu/openpilot/commit/cc30adabbbb3c3eebbeda10907bdcc3cce7e01cf) to be able to connect to my WPA3 SSID but it doesn't work:
```
selfdrive/ui/qt/network/wifi_manager.cc: DBus call error: "Did not receive a reply. Possible causes include: the remote
 application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the
 network connection was broken."
```
also tried with nmcli [1.46.0](https://packages.ubuntu.com/noble/network-manager) in Ubuntu 24.04:
```
nmcli -v
nmcli tool, version 1.46.0

nmcli conn show
NAME                           UUID                                  TYPE      DEVICE
...
openpilot connection **redacted-ssid**  e5aea78d-42bb-4082-b4b8-d978c24098c9  wifi      --

ifconfig wlan0
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:0a:f5:90:e8:d4  txqueuelen 3000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

dmesg -w
...
[ 1819.334487] R0: [soft_irq][35058324353] [09:07:21.967314]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[ 1819.339776] R0: [soft_irq][35058426523] [09:07:21.972635]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[ 1836.334879] R0: [cds_mc_thread][35384732358] [09:07:38.967482]  wlan: [45934:E :SME] csr_get_active_scan_entry: 4978: Active list Empty scanId: 0
[ 1845.045532] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

<details>
  <summary>compared to a connecting to a WPA2 wifi afterwards</summary>

```
[ 1882.834414] R0: [soft_irq][36277523518] [09:08:25.466340]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[ 1882.839890] R0: [soft_irq][36277628633] [09:08:25.471815]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[ 1882.903422] R0: [wpa_supplicant][36278848348] [09:08:25.535340]  wlan: [3591:I :HDD] hdd_sme_roam_callback: 5141: Disabling queues
[ 1882.903621] R0: [wpa_supplicant][36278852290] [09:08:25.535546]  wlan: [3591:E :SME] csr_send_join_req_msg: 15855: Connecting to ssid:systeam bssid: 5c:02:14:b0:21:e2 rssi: -46 channel: 1 country_code: RO
[ 1882.905370] R0: [cds_mc_thread][36278885837] [09:08:25.537293]  wlan: [45934:I :WMI] send_vdev_start_cmd_tlv: vdev_id 0 freq 2412 chanmode 5 ch_info: 0x5 is_dfs 0 beacon interval 100 dtim 1 center_chan 2412 center_freq2 0 reg_info_1: 0x140000 reg_info_2: 0x1400, req->max_txpow: 0x14 Tx SS 2, Rx SS 2, ldpc_rx: 0
[ 1883.037358] R0: [cds_mc_thread][36281420020] [09:08:25.669280]  wlan: [45934:I :WMA] wma_send_peer_assoc: vdev_id 0 associd 1 peer_flags 219006 nss 2 phymode 5 ht_caps 1ed
[ 1883.037372] R0: [soft_irq][36281420316] [09:08:25.669295]  wlan: [0:I :TRX] ol_txrx_enqueue_rx_frames: 5519: Data on the peer before it is registered bufq->curr 0 bufq->drops 0
[ 1883.044946] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1883.067431] R0: [soft_irq][36281997446] [09:08:25.699354]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[ 1883.072234] wlan: 00:0a:f5:90:e8:d4 connected to 5c:02:14:b0:21:e2
[ 1883.072298] R0: [cds_mc_thread][36282090912] [09:08:25.704221]  wlan: [45934:I :HDD] hdd_send_peer_status_ind_to_oem_app: 716: OEM app is not registered(0) or pid is invalid(0)
[ 1883.109110] R0: [wpa_supplicant][36282797636] [09:08:25.741030]  wlan: [3591:I :HDD] hdd_print_netdev_txq_status: 18808: netdev tx queue[0] state: 0x0
[ 1883.109134] R0: [wpa_supplicant][36282798178] [09:08:25.741058]  wlan: [3591:I :HDD] hdd_print_netdev_txq_status: 18808: netdev tx queue[1] state: 0x0
[ 1883.109154] R0: [wpa_supplicant][36282798556] [09:08:25.741077]  wlan: [3591:I :HDD] hdd_print_netdev_txq_status: 18808: netdev tx queue[2] state: 0x0
[ 1883.109204] R0: [wpa_supplicant][36282798925] [09:08:25.741097]  wlan: [3591:I :HDD] hdd_print_netdev_txq_status: 18808: netdev tx queue[3] state: 0x0
[ 1883.109224] R0: [wpa_supplicant][36282799904] [09:08:25.741148]  wlan: [3591:I :HDD] hdd_print_netdev_txq_status: 18808: netdev tx queue[4] state: 0x0
[ 1883.109284] R0: [wpa_supplicant][36282801045] [09:08:25.741207]  wlan: [3591:I :HDD] __wlan_hdd_cfg80211_disconnect: 18909: Disconnect request from user space with reason: 2 (WLAN_REASON_PREV_AUTH_NOT_VALID) internal reason code: 3
[ 1883.109334] R0: [wpa_supplicant][36282802027] [09:08:25.741258]  wlan: [3591:I :HDD] wlan_hdd_disconnect: 18665: Disabling queues
[ 1883.114688] R0: [cds_mc_thread][36282904744] [09:08:25.746608]  wlan: [45934:W :WMA] wma_delete_bss: 5564: Outstanding msdu packets: 0
[ 1883.114861] R0: [cds_mc_thread][36282908133] [09:08:25.746785]  wlan: [45934:E :WMA] Invalid wma_cli_set vdev command/Not yet implemented 0x75
[ 1883.128876] R0: [cds_mc_thread][36283177186] [09:08:25.760797]  wlan: [45934:I :WMA] wma_remove_peer_by_reference: 1846: Deleting peer 5c:02:14:b0:21:e2 vdev id 0
[ 1883.128984] R0: [cds_mc_thread][36283179280] [09:08:25.760907]  wlan: [45934:I :TRX] ol_txrx_peer_detach:peer fffffff77fedb800 (5c:02:14:b0:21:e2)
[ 1883.144061] R0: [cds_mc_thread][36283468761] [09:08:25.775983]  wlan: [45934:E :PE ] pe_delete_session: 683: session is not valid
[ 1883.144316] R0: [cds_mc_thread][36283473675] [09:08:25.776239]  wlan: [45934:I :HDD] hdd_dis_connect_handler: 1651: Disabling queues
[... cut ...]
[ 1901.029431] R0: [NetworkManager][36626867499] [09:08:43.661072]  wlan: [3560:E :HDD] hdd_is_rcpi_applicable: 4948: mac addr is different from bssid connected
[ 1907.029880] R0: [NetworkManager][36742075895] [09:08:49.661420]  wlan: [3560:E :HDD] hdd_is_rcpi_applicable: 4948: mac addr is different from bssid connected
[ 1913.029222] R0: [NetworkManager][36857263587] [09:08:55.660690]  wlan: [3560:E :HDD] hdd_is_rcpi_applicable: 4948: mac addr is different from bssid connected
```

</details>

So, it still doesn't work also in Ubuntu 24.04 with Network Manager 1.46.0

andiradulescu, 2025-02-27:
From what I remember it will work only with kernel driver support. Or when we switch to mainline.

shunnag, 2026-09-24:
Root cause found and fixed in the kernel; WPA3-only networks now connect on a comma four (AGNOS 19.8) with a 4-file qcacld-3.0 change. Details and on-device logs below.

## Root cause

`drivers/staging/qcacld-3.0/Kbuild` only sets `CONFIG_WLAN_FEATURE_SAE := y` inside `ifeq ($(KERNEL_BUILD), 0)`, the Android external-module path (`MODNAME=wlan`). AGNOS builds qcacld in-tree (`CONFIG_QCA_CLD_WLAN=y`, `MODNAME` empty → `KERNEL_BUILD=1`), and neither the qcacld Kconfig nor `tici_defconfig` has an SAE symbol. So `-DWLAN_FEATURE_SAE` has never been defined in any AGNOS kernel since the 2020 import (commaai/agnos-kernel-sdm845@2c6aed22f9ff).

As a result, the driver never sets `NL80211_FEATURE_SAE`, and wpa_supplicant drops SAE before it sends CONNECT. This explains the missing `csr_send_join_req_msg` in the earlier dmesg and NM's `ssid-not-found`. It is not a NetworkManager / Ubuntu / firmware issue.

The shipped 19.8 `boot.img` (335c1757…) contains none of the SAE-only log strings, and its `.external_auth` cfg80211 op is NULL.

## Before (stock AGNOS 19.8, comma four, WPA3-only AP: SAE, PMF required, no H2E)

```
$ wpa_cli -i wlan0 get_capability auth_alg
OPEN SHARED LEAP
NetworkManager: Config: added 'key_mgmt' value 'SAE FT-SAE'
wpa_supplicant: WPA: AP key_mgmt 0x400 network profile key_mgmt 0xc00; available key_mgmt 0x0
wpa_supplicant: WPA: Failed to select authenticated key management type
NetworkManager: Activation: (wifi) association took too long, failing activation
$ nmcli dev wifi connect <ssid> password <pw>
Error: Connection activation failed: The Wi-Fi network could not be found.
```

## Fix

Kernel PR: commaai/agnos-kernel-sdm845#142. It changes 4 files (+11/−2):

1. Add a `WLAN_FEATURE_SAE` Kconfig symbol (like `WLAN_FEATURE_11W`) and set `CONFIG_WLAN_FEATURE_SAE=y` in `tici_defconfig`.
2. Move `case NL80211_AUTHTYPE_SAE` in `wlan_hdd_cfg80211_set_auth_type()` out of the `WLAN_FEATURE_FILS_SK` guard. FILS is off in-tree, so enabling SAE alone still fails with `-EINVAL`. This matches the newer 5.1.1.75T driver in commaai/agnos-kernel-sdm845#98.
3. Port the "SAE with open authentication" branch in `hdd_set_csr_auth_type()` from the commaai/agnos-kernel-sdm845#98 driver. Without it, SAE AKM + `NL80211_AUTHTYPE_OPEN_SYSTEM` (sent for PMKSA-caching reconnects, and by NM profiles with `auth-alg=open`) is mapped to `eCSR_AUTH_TYPE_RSN`.

Build: agnos19.8 branch, kernel eccd146, comma's GCC 8.2.1 toolchain. A control build without the patch matches the shipped image except for build identity and module-signing key. With the patch, only `drivers/staging/qcacld-3.0` (plus `init/version.o` and `kernel/configs.o`) is rebuilt. The added strings are all SAE-related.

## After (same device, 19.8 + patched boot.img)

```
$ wpa_cli -i wlan0 get_capability auth_alg
OPEN SHARED LEAP SAE
$ nmcli dev wifi connect <ssid> password <pw>
Device 'wlan0' successfully activated
nl80211: BSS Event 127 (NL80211_CMD_EXTERNAL_AUTH) received for wlan0
SAE: Selecting supported ECC group 19
wlan0: nl80211: External auth status: 0
wlan0: WPA: Key negotiation completed with <bssid> [PTK=CCMP GTK=CCMP]
# DHCP ok, ping ok; BIP (AES-128-CMAC) negotiated
$ wpa_cli -i wlan0 reassociate          # PMKSA caching path (fix 3)
wlan0: SAE: Use WPA_AUTH_ALG_OPEN for PMKSA caching attempt
wlan0: CTRL-EVENT-CONNECTED
```

The stock WLAN firmware (`WLAN.HL.2.0.c10-00459`, 2019) handles external-auth SAE and PMF fine.

The driver skips firmware roam offload for SAE connections (`Roaming not supported for SAE connection`, by design in this driver). SAE links reconnect instead of roaming.

## Remaining / notes

- **openpilot UI:** `get_security_type()` still marks SAE-only APs `UNSUPPORTED` and always creates `key-mgmt=wpa-psk` profiles.
  - Ubuntu's wpa_supplicant 2.10 doesn't report `sae` in D-Bus `KeyMgmt` for this driver, because the driver advertises no AKM suites (fixed upstream in hostap 2.12, d82c9e61). So NM won't add SAE to `wpa-psk` profiles on its own.
  - The UI needs to create `key-mgmt=sae` profiles (without `auth-alg`) for SAE-only APs, gated on `GET_CAPABILITY auth_alg` containing `SAE`. I have a patch for that, tested on the same device: a WPA3-only AP shows with a lock in the mici UI, and connecting creates a `sae` profile (`key_mgmt SAE FT-SAE`, `ieee80211w 2`, no `auth_alg`). Saved-profile reconnect works. A wrong password ends in `need-auth` → `failed (no-secrets)` after about 26 s, and the UI handles it. Transition APs still use `wpa-psk`. I'll open that PR after aligning with the NM-removal work (commaai/openpilot#37752 / commaai/openpilot#38968), since the wpa_supplicant-direct path needs the same change.
  - WPA2/WPA3 transition networks keep using WPA2-PSK.
- **Not supported:** the driver has no RSNXE/SAE-H2E code, so H2E-only APs (sae_pwe=1, e.g. 6 GHz-style configs) will likely still fail. Hunting-and-pecking works.
- **Not tested:** comma 3X. It uses the same kernel and firmware.

Full before/after logs (redacted): https://gist.github.com/shunnag/90458a4e96a36e6260145a9ab1226edd

shunnag, 2026-09-25:
Follow-up for SAE hash-to-element (H2E), which my earlier comment listed as unsupported:

- commaai/agnos-kernel-sdm845#143 (draft, on top of #142): qcacld-3.0 now puts the RSNXE from wpa_supplicant in the association request. It ports upstream qcacld-3.0 c051588e6, dfba36b6f and 7c07f228f6.
- #629: rebuilds Ubuntu's wpa_supplicant 2.10 with `sae_pwe=2` as the default and the SAE fixes from hostap 2.11/2.12 (advisories 2024-2, 2026-2, 2026-3). NetworkManager behaves as before: transition networks stay WPA2-PSK and the hotspot is unchanged.

Tested on a comma four against a Wi-Fi 7 mesh AP in WPA3-only mode that advertises H2E:

- H2E connects (status 126), and RSNXE is in the association request (`RSN: set own RSNXE` instead of `clearing own RSNXE`).
- PMKSA-cached reconnect and wrong-password handling work.
- The stock WLAN firmware relays the H2E auth frames without changes.

My AP supports both SAE methods, so an H2E-only AP is not tested on the device yet. It connects on mac80211_hwsim with hostapd `sae_pwe=1`.

adeebshihadeh, 2026-09-26:
Thanks, this is good to know. Can you port your changes over to our new OS, [vamOS](https://github.com/commaai/vamOS)? AGNOS is currently in maintenance mode.

shunnag, 2026-09-26:
Thanks for taking a look! I'll see if I can port my changes to vamOS.

shunnag, 2026-09-26:
I tried vamOS on my comma four. WPA3-Personal works there without porting anything, but I found one reconnect problem with PMF.

**Setup:**
- Liberation Day 7.2 (`vamos.json` 442f27c3), installed into the inactive slot the way dorapilot does it.
- AGNOS 19.8 bootloaders: xbl e8acf2a9, abl 29fd7ed1.
- WCN3990 firmware WLAN.HL.2.0.c10-00459.
- wpa_supplicant 2.12, NetworkManager 1.56.
- The first attempt ended in EDL (9008) about 1.4 s after the reset, before ABL picked a slot (boot_a retry count unchanged). A cold boot from port 2 with the PC unplugged booted LD normally.

**Nothing from the AGNOS work needs porting.** commaai/agnos-kernel-sdm845#142, commaai/agnos-kernel-sdm845#143 and #629 are qcacld- and Ubuntu-2.10-specific. With mac80211, wpa_supplicant does SAE itself, and NetworkManager sets `SaePwe=2`.

**What works on vamOS:**
- **WPA3-only SSID** (TP-Link Deco BE85, SAE + PMF required): connects with H2E (status 126, `sae_h2e=1`, RSNXE `f4 01 20`). DHCP and ping work.
- **WPA2/WPA3 transition SSIDs**: connect with SAE through an ordinary `wpa-psk` profile, because NM adds SAE. Tested on the Deco and on an iPhone hotspot.
- **`mfp` feature bit** (see samcday/pocketfed#27): not needed on this firmware. PMF links pass data with the stock `firmware-5.bin`. Setting the bit changed nothing I could see.
- **SAE hotspot**: hostapd 2.11 run directly works. An iPhone joined with AKM 00-0f-ac:8, MFP and DHCP. NetworkManager's AP mode failed with EBUSY when it changed the interface type; hostapd got past the same EBUSY by taking the interface down first. Filed as commaai/vamOS#132.

**Problem: reconnecting after a PMF disconnect** (filed as commaai/vamOS#131).
- On the Deco (Wi-Fi 7, MLO always on), after vamOS disconnects a PMF connection, an immediate reconnect fails for about 40 s to 2 min.
  - The AP answers with association comeback, the association times out, and auth frames from that MAC then get no answer for a while.
  - This happened with SAE and with WPA2-PSK + PMF. WPA2-PSK without PMF reconnects in about 1 s.
  - With a different MAC address the reconnect works in 3 s. So the AP still thinks the old PMF association is alive.
- On the same device, AGNOS (qcacld) reconnects to the same SSID in 0-1 s (3/3). It looks like the Deco does not accept the deauth that vamOS sends when it disconnects a PMF link. I have no over-the-air capture to show why.
- An iPhone hotspot (no MLO) mostly reconnected right away.

**openpilot side:**
- WPA3-only networks still need a `key-mgmt=sae` profile.
- The UI patch I have checks wpa_supplicant's control socket, which vamOS doesn't create. Since commaai/openpilot#38968 replaces NetworkManager, SAE probably belongs there: `key_mgmt SAE`, `ieee80211w`, `sae_password`, global `sae_pwe=2`.

**Side note on recovery:** official qdl.js `setactiveslot` changes only the slot attributes and leaves the partition type GUIDs on the other slot. ABL then refused to boot ('press any key to shutdown') until I restored the GPT. andiradulescu/qdl.js swaps the GUIDs.

I can share redacted logs if they help.

shunnag, 2026-09-26:
Correction to my comment above: I wrote that the `mfp` feature bit (samcday/pocketfed#27) was not needed on this firmware. That was wrong.

Without the bit, data frames work, but ath10k never marks PMF peers to the firmware. The firmware then sends the protected Deauth with PN 0, the AP drops it, and that causes the reconnect problem in commaai/vamOS#131.

With the bit set, 6 of 7 reconnects took 1-5 s instead of 21-25 s. Captures and details are in commaai/vamOS#131, and the proposed fix is commaai/vamOS#133.

shunnag, 2026-10-01:
Follow-up for the record (no action needed, since AGNOS is in maintenance mode).

I ran the same measurements on AGNOS 19.8 with the WPA3 kernel and wpa_supplicant from my fork, on the same comma four and AP (5 GHz ch 48, iperf3 30 s each way, profiles created by the openpilot UI):

| | AGNOS 19.8 + WPA3 | vamOS LD 7.2 (stock) |
|---|---|---|
| WPA3-only (SAE, PMF required), Mbit/s up / down | 259 / 228, 270 / 227 | 32 / 22, 23 / 22 |
| WPA2/WPA3 mixed | 271 / 217 (WPA2-PSK + PMF) | 23 / 23 (SAE + PMF) |
| WPA2-only | 268 / 266, 270 / 262 | 283 / 265, 280 / 272 |
| Reconnect, same MAC | 0.8-1.4 s | 24 s (status 30) |
| Connect right after a failed attempt | 1.5 s (3/3, wrong SAE password first) | ~20 s firmware Auth hold |

So the slow PMF links and the reconnect delays only happen on vamOS. qcacld sets the peer PMF flag and cleans up after a failed attempt; ath10k does not. The vamOS side is tracked in commaai/vamOS#131 and #133.

The run log, the iperf3 per-second summary and the idle ping results are attached. SSIDs and AP addresses are removed.

Note: I worked through this together with Claude (AI). English is hard for me, so the English was written with AI help. I ran the device tests myself on my comma four, and the results above come from those runs.

[agnos_wpa3_measurements.txt](https://github.com/user-attachments/files/32881592/agnos_wpa3_measurements.txt)