commaai/agnos-builder #187: Devices Cgroup Support

MisterBerg, 2024-01-01:
I am using the Comma3x dev kit for various work and personal purposes. I would like to use Docker as an easy way to deploy and test new software on the device without installing dependencies which might break the existing software.environment.

Devices cgroup kernal support is a requirement for Docker.  I rebuilt the kernel to add this support, but the devices cgroup somehow breaks WiFi module functionality.  

I could dive into an investigation, but I thought I would ask whether the Comma.ai team has been down this road already.  If so, how far did you go? Is it worth my time to dig in, or am I going to discover some limitation that I can't really get around?

Thanks in advance for any directional support you might be able to offer!

MisterBerg, 2024-01-04:
@adeebshihadeh 

Sorry to ping you personally, but I can see that you are the most active comma.ai member maintaining this repo!

I'm really just looking for some quick info if you have it.  Feel free to tell me that the team has never tried to use docker with the Comma3(x), or to just buzz off entirely :)

Thanks!

adeebshihadeh, 2024-01-04:
Nope, never tried. Gonna close this since we have no plans to support it, but we'd probably merge a straightforward PR to add support for if it doesn't break anything or regress boot time.

closed 2024-01-04 by adeebshihadeh

MisterBerg, 2024-01-04:
Thanks for the reply, Adeeb

I'll look into it and see how difficult it might be.
It's a really cool and exciting product hopefully I can contribute in some
way.

Have a good week!

adeebshihadeh, 2024-01-04:
Feel free to try this, but we really can't spend much time validating and merging stuff that isn't relevant to solving self driving cars.

If you want to contribute to that goal, there's lots of fun bounties up: https://github.com/orgs/commaai/projects/26/views/1. Making it boot faster is a good one for AGNOS.