commaai/agnos-builder #209: SSH should start without weston running

andiradulescu, 2024-02-06:
If weston crashes upfront (before init), e.g. `/usr/bin/weston: error while loading shared libraries: libffi.so.6`, then SSH doesn't start.

If weston crashes later on (after init), then SSH starts.

My assumption is that in the first case `weston-ready.service` is not started, `comma.sh` doesn't do the right thing (?) and `continue.sh` is not called, thus `SshEnabled` is not being touched, thus `ssh-param-watcher.path` is not being triggered. Didn't investigate properly, just a hunch.

adeebshihadeh, 2024-02-28:
I believe this works now?

andiradulescu, 2024-02-28:
If you didn't do any changes to `ssh-param-watcher.path` and `weston-ready.service` then it will still not work.

As I explained, my assumption is it happens only when `weston-ready.service` is not getting started. When this happens, `continue.sh` is not called, so `ssh-param-watcher.path` isn't getting triggered.

_Since openpilot doesn't start and nobody else touches `SshEnabled` so to trigger `ssh-param-watcher.path`._

One way of resolving this is by removing all moving parts, and interact directly with `systemd` via `systemctl` to enable/disable the `ssh` service.