commaai/agnos-builder #230: [$250 bounty] System image and kernel building in CI

adeebshihadeh, 2024-06-19:
Ideally this runs on an arm64 runner - not sure if GHA offers those on the open source tier though.

Requirements
* `build_system.sh` and `build_kernel.sh` running in CI
* uploads the built images as an [artifact](https://docs.github.com/en/actions/using-workflows/storing-workflow-data-as-artifacts)
* must have caching between runs so the avg run is (somewhat) fast

andiradulescu, 2024-06-20:
Regarding `arm64` runners on open source tier:

> These runners are available to our customers on our GitHub Team and Enterprise Cloud plans. We expect to begin offering Arm runners for open source projects by the end of the year.

> Customers can begin using these runners today by creating an Arm runner in their organization/enterprise, and then updating the runs-on syntax in their GitHub Actions workflow file to call that runner name.

> Find out more information on how to set up Arm-hosted runners in [this video](https://www.youtube.com/watch?v=cgI6SBP8pEM) or in our [public documentation](https://docs.github.com/actions/using-github-hosted-runners/about-larger-runners/about-larger-runners).

https://github.blog/2024-06-03-arm64-on-github-actions-powering-faster-more-efficient-build-systems/#get-started-using-arm-hosted-runners-today

andiradulescu, 2024-07-16:
Incremental caching and system build with https://github.com/commaai/agnos-builder/pull/246 as soon as it's ready for review

PR #231 (2024-07-14): Build kernel in CI
- [x] build kernel
  - [x] use ccache (seems to reduce from 13 -> 3 mins)
  - [x] upload `boot.img` artifact
- [x] build system
  - [x] switch to arm64 custom runner
  - [x] pull/cache docker steps (need to wait for successful run?)

closes #230
closed 2024-07-21 by adeebshihadeh