commaai/agnos-builder #240: Switch from pyenv to uv

adeebshihadeh, 2024-07-14:
Similar to #232, but for managing the Python installation

eshaanrathi2, 2024-07-17:
@adeebshihadeh uv is a replacement for pip. i don't think it is equivalent for pyenv.
though, would it help if I migrate pip package installs to uv, perhaps making it faster?

adeebshihadeh, 2024-07-17:
```bash
batman:~$ uv python list
warning: `uv python list` is experimental and may change without warning.
cpython-3.12.4-linux-x86_64-gnu     <download available>
cpython-3.11.9-linux-x86_64-gnu     <download available>
cpython-3.11.4-linux-x86_64-gnu     .pyenv/versions/3.11.4/bin/python3
cpython-3.11.4-linux-x86_64-gnu     .pyenv/versions/3.11.4/bin/python
cpython-3.10.14-linux-x86_64-gnu    <download available>
cpython-3.9.19-linux-x86_64-gnu     <download available>
cpython-3.8.19-linux-x86_64-gnu     <download available>
cpython-3.8.10-linux-x86_64-gnu     /usr/bin/python3
cpython-3.8.10-linux-x86_64-gnu     /bin/python3
cpython-3.7.9-linux-x86_64-gnu      <download available>
batman:~$ uv python install 3.12
warning: `uv python install` is experimental and may change without warning.
Searching for Python versions matching: Python 3.12
Found existing installation for Python 3.12: cpython-3.12.3-linux-x86_64-gnu
batman:~$
```

coder351, 2024-07-17:
@adeebshihadeh I have prepared PR(https://github.com/commaai/agnos-builder/pull/248) which switches pyenv to uv for openpilot_python_dependencies.sh.

pantew869, 2024-07-18:
I've spent a couple of hours on this, dbus-python fails to build because meson can't find pkg-config files for python, and even if you fix this, the pkg-config files provided by the uv have incorrect paths in them.

closed 2024-09-03 by adeebshihadeh