commaai/agnos-builder #297: Error on docker warnings in CI

adeebshihadeh, 2024-08-09:


robin-reckmann, 2024-08-09:
Functionality is implemented: https://github.com/commaai/agnos-builder/pull/298
Test: https://github.com/commaai/agnos-builder/pull/299

It works locally, but the behavior in CI is different. Warning is displayed but no error. Since build checks is a new feature (see https://docs.docker.com/build/checks/), he CI docker version might be slightly outdated.

robin-reckmann, 2024-08-09:
Local result:
```WARNING: NoEmptyContinuation - https://docs.docker.com/go/dockerfile/rule/no-empty-continuation/
Empty continuation line
Dockerfile.agnos:216
--------------------
 214 |         rm -rf /root/.cache && \
 215 |         
 216 | >>>     apt-get clean
 217 |     
--------------------

ERROR: lint violation found for rules: NoEmptyContinuation```

CI result:
```NoEmptyContinuation - https://docs.docker.com/go/dockerfile/rule/no-empty-continuation/
Empty continuation lines will become errors in a future release
Dockerfile.agnos:216
--------------------
 214 |         rm -rf /root/.cache && \
 215 |         
 216 | >>>     apt-get clean
 217 |     
--------------------```

closed 2024-08-11 by adeebshihadeh