commaai/agnos-builder #319: Extra serial output after login is up

adeebshihadeh, 2024-08-20:
Seems to be a recent regression (24.04?). On AGNOS 10, there's no output once the the login prompt is up.

```
[    8.669750] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS2
[    8.677948] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS2 -> direct -> AMIC2
[    8.682400] wlan: Loading driver v5.1.1.63E ( +PANIC_ON_BUG)
[    8.690282] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister lo
[    8.700654] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS2
[    8.708610] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Headset Mic -> direct -> MIC BIAS2
[    8.718345] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS2
[    8.726478] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS2 -> direct -> AMIC3
[    8.735684] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS2
[    8.743738] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route ANCRight Headset Mic -> direct -> MIC BIAS2
[    8.754283] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS2
[    8.762446] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS2 -> direct -> AMIC4
[    8.771649] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS2
[    8.779546] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route ANCLeft Headset Mic -> direct -> MIC BIAS2
[    8.789989] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS3
[    8.798193] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS3 -> direct -> AMIC5
[    8.807533] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS3
[    8.815515] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Handset Mic -> direct -> MIC BIAS3
[    8.825220] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS1
[    8.833347] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS1 -> direct -> DMIC0
[    8.842542] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS1
[    8.850487] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic0 -> direct -> MIC BIAS1
[    8.860313] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS1
[    8.868419] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS1 -> direct -> DMIC1
[    8.877631] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS1
[    8.885669] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic1 -> direct -> MIC BIAS1
[    8.895487] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS3
[    8.903647] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS3 -> direct -> DMIC2
[    8.912838] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS3
[    8.920881] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic2 -> direct -> MIC BIAS3
[    8.930698] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS3
[    8.938832] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS3 -> direct -> DMIC3
[    8.948096] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS3
[    8.956065] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic3 -> direct -> MIC BIAS3
[    8.965951] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS4
[    8.974125] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS4 -> direct -> DMIC4
[    8.983409] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS4
[    8.991360] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic4 -> direct -> MIC BIAS4
[    9.001189] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for MIC BIAS4
[    9.009309] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route MIC BIAS4 -> direct -> DMIC5
[    9.018506] sdm845-asoc-snd soc:sound-tavil: ASoC: no sink widget found for MIC BIAS4
[    9.026452] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route Digital Mic5 -> direct -> MIC BIAS4
[    9.036294] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for SPK1 OUT
[    9.044484] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route SPK1 OUT -> direct -> SpkrLeft IN
[    9.054189] sdm845-asoc-snd soc:sound-tavil: ASoC: no source widget found for SPK2 OUT
[    9.062196] sdm845-asoc-snd soc:sound-tavil: ASoC: Failed to add route SPK2 OUT -> direct -> SpkrRight IN
[    9.072185] max98088 0-0010: ASoC: mux External MIC has no paths

Ubuntu 24.04 LTS comma-c9b8599a ttyMSM0

comma-c9b8599a login:
comma-c9b8599a login: comma
Password:
                      .~ssos+.
                    +8888888888i,
                   {888888888888o.
                   h8888888888888k
                   t888888888s888k
                    `t88888d/ h88k
                       ```    h88l
                             ,88k`
                            .d8h`
                           +d8h
                        _+d8h`
                      ;y8h+`
                      |-`

[... 42 lines cut ...]
[   23.155226] rmt_storage:INFO:rmt_storage_client_thread: Calling Read [offset=1024, size=2096128]for /boot/modem_fs2!
[   23.181711] rmt_storage:INFO:rmt_storage_client_thread: Done Read (bytes = 2096128) for /boot/modem_fs2!
[   23.345036] diag: In diag_send_feature_mask_update, control channel is not open, p: 0, ffffff951132b478
[   23.358144] diag: In diag_send_peripheral_buffering_mode, buffering flag not set for 0
[   24.809849] IPC_RTR: process_new_server_msg: Server 00001003 create rejected, version = 0
[   28.429885] ipa ipa3_uc_reg_rdyCB:1727 bad parm. inout=          (null) [   28.441599] ipa ipa3_uc_reg_rdyCB:1727 bad parm. inout=          (null)
[   28.468421] ipa ipa3_uc_reg_rdyCB:1727 bad parm. inout=          (null) [   28.549119] IPC_RTR: process_new_server_msg: Server 00001003 create rejected, version = 0
[   28.646145] cnss_utils: WLAN MAC address is not set, type 0
[   39.343507] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip6tnl0
[   39.390458] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip6tnl0
[   39.399156] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister sit0
[   39.420283] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister sit0
[   39.428587] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip6_vti0
[   39.450272] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip6_vti0
[   39.458716] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip_vti0
[   39.500332] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister ip_vti0
[   39.509032] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister lo
[   39.540310] [RMNET:HI] rmnet_config_notify_cb(): Kernel is trying to unregister lo
[  113.808769] rmt_storage:INFO:rmt_storage_rw_iovec_cb: Write iovec request received for /boot/modem_fs1
[  113.820776] rmt_storage:INFO:rmt_storage_client_thread: Calling Write [offset=0, size=2097152]for /boot/modem_fs1!
[  113.901102] rmt_storage:INFO:rmt_storage_client_thread: Done Write (bytes = 2097152) for /boot/modem_fs1!

comma@comma-c9b8599a:/data/openpilot$
comma@comma-c9b8599a:/data/openpilot$
```

robin-reckmann, 2024-08-21:
Just to confirm, this is console output on ttyMSM0, or?

robin-reckmann, 2024-08-21:
Would you like to keep the different behavior between before and after login? It seems like all kernel prints are suppressed from being output to console. I did some investigation but couldn't finally figure out what exactly cause this in agnos<11. If we are fine with keeping the intended behavior of logging kernel messages of a certain loglevel to console we could modify /etc/sysctl.d/10-console-messages.conf to only print message with log level higher than error. This would suppress all messages in the current build (both before and after login is up). They can still of course be checked with dmesg.

robin-reckmann, 2024-08-26:
I created a PR to fix this. I couldn't find the exact reason for this behavior, but using systemd 255 with the outdated 4.9 kernel might lead to this kind of issues. I propose to merge this fix for now and see if we can revert it when updating the kernel.

PR #326 (2024-08-27): Hang up console before serial-getty start
In Ubuntu 24.04 (systemd 255) and Kernel 4.9 the boot console is not properly hung up before the start of serial-getty@ttyMSM0. In this state, kernel messages will continue to be printed to the serial console.
By sending another hupcl (Hang Up on Close) using stty, we can force the hangup and kernel messages will be properly handled by systemd and written to the appropriate log files without appearing on screen. 
This fixes #319.
closed 2024-08-27 by adeebshihadeh