commaai/agnos-builder #321: AGNOS 11

adeebshihadeh, 2024-08-20:
Tracking issue for the AGNOS 11 release.

AGNOS 11 is:
- [x] Ubuntu 24.04
- [ ] optimized `uv python`
- [ ] [ABL without](https://github.com/commaai/agnos-builder/issues/320)

andiradulescu, 2024-09-03:
@adeebshihadeh for a shipping bounty, I believe I can make AGNOS 11 shippable by Sep 30.

1. https://github.com/commaai/agnos-builder/issues/225 - 1 week for a small image - probably close to 2 weeks for a very small image - also includes:

- [ ] https://github.com/commaai/agnos-builder/issues/329
- [ ] https://github.com/commaai/agnos-builder/issues/23
- [ ] https://github.com/commaai/agnos-builder/issues/335

2. https://github.com/commaai/agnos-builder/issues/320 - 1 week minimum, probably close to 2 weeks - includes:

- [ ] https://github.com/commaai/flash/pull/39
- [ ] https://github.com/commaai/flash/issues/53
- [ ] https://github.com/commaai/agnos-builder/issues/337

3. Smaller tasks that also need to be ready to ship - a couple of days - includes (and may include other small things also):

- [ ] https://github.com/commaai/agnos-builder/issues/301
- [ ] https://github.com/commaai/agnos-builder/issues/318
- [ ] https://github.com/commaai/agnos-builder/issues/240

adeebshihadeh, 2024-09-04:
I think this can be done quite a bit faster. #1 and #2 are practically the same thing. Here's how I'd approach it in order:
* Figure out how small the image needs to be to be <= fastboot flashing time (over USB 3), 1-3 hours
* Reduce image size exactly that much, ~2 days (maybe less if we don't need to 
reduce much or even just something about the flashing)
* Verify and merge https://github.com/commaai/flash/pull/39, 1 day
* https://github.com/commaai/agnos-builder/issues/337, 10 mins

All the #3 things are small. I think they can be done in parallel while waiting for stuff to build and not take up much time.  

I see you added #329 to the milestones, but I think that confuses things. The focus for v11 is 24.04 + saving the 3s from no fastboot. We can get the cleanups in when they fit in, but otherwise they're not a priority.

The <1GB target in https://github.com/commaai/agnos-builder/issues/225 is a bit arbitrary. As long as all of AGNOS flashes in <1min over USB 3, I think we're good.

andiradulescu, 2024-09-04:
Actually I added https://github.com/commaai/agnos-builder/issues/329 since it was in milestone added by you before. Let’s remove it.

Keeping in mind what you said, AGNOS 11 should be shippable by end of this week. Without cleanups and just get QDL shipped basically.

As for reducing the image, the “splits” above should get the image at the “proper” size.

As for estimates, no idea I can ship QDL in just one day of work, I need to start working on it first.

adeebshihadeh, 2024-09-16:
Issues found when installing on CI devices
- [ ] missing dev dependencies
- [ ] new sentry sdk version isn't supported in openpilot yet
- [ ] https://github.com/commaai/agnos-builder/issues/374

andiradulescu, 2024-09-17:
Regarding Sentry, this is not a bug, `sentry-sdk` got updated to `2.x` so the sentry implementation needs to be updated - I traced `sentry-sdk` as back as moving to uv (still `2.x` then):

Although I have previous experience with Sentry, I can't straight figure out what this [line](https://github.com/commaai/openpilot/blob/23716986011814e4ef1c4ea72c5e2d2afc8184b0/system/sentry.py#L76) is doing or better said how should it be migrated to [2.x](https://docs.sentry.io/platforms/python/migration/1.x-to-2.x) - since it's not cloning the hub as stated [here](https://docs.sentry.io/platforms/python/migration/1.x-to-2.x#activating-current-hub-clone).

The deprecated [function](https://github.com/getsentry/sentry-python/blob/4f6ccc45af0e0cc0a09f3b38c76f02b49e469feb/sentry_sdk/hub.py#L600) does this:
```
    def start_session(self, session_mode="application"):
        get_isolation_scope().start_session(
            session_mode=session_mode,
        )
```

Compared to sentry [1.x](https://github.com/getsentry/sentry-python/blob/282b8f7fae3da3c3ec26e5ee5e1599fc74661a72/sentry_sdk/hub.py#L619):
```
    def start_session(self, session_mode="application"):
        client, scope = self._stack[-1]
        scope.start_session(
            client=client,
            session_mode=session_mode,
        )
```

@adeebshihadeh can you take care of this or should I? Might take me more time than expected to figure everything out and be sure any changes I make are entirely correct.

closed 2024-10-31 by adeebshihadeh