commaai/agnos-builder #325: dmesg cleanup project

adeebshihadeh, 2024-08-24:
We've got a bunch of minor warnings and errors reported in dmesg that are expected. We should fix those up, then we can start scanning `dmesg` to mine for long tail issues, such as UFS failures.

adeebshihadeh, 2024-08-27:
On second thought, maybe we should just be strict about this when we do mainline. @andiradulescu if you agree, let's close

andiradulescu, 2024-08-28:
as a more generic answer, I would not spend time on things that also need to be checked/fixed on mainline, unless we want to release AGNOS 11 with the old kernel - which I know we don't.

so, let's leave this issue for (after) mainline, since on mainline I also saw some dmesg warnings/errors that should be fixed. this issue will remind us to not leave any warnings (even small) in AGNOS 11.

adeebshihadeh, 2024-08-31:
Added a "Validation" section to your top level comment on the mainline PR

closed 2024-08-31 by adeebshihadeh