commaai/agnos-builder #370: build_system.sh fails with code 100 during compile-qtwayland5.sh

MisterBerg, 2024-09-15:
It has been a long time since I tried to use the Agnos-Builder tool.
The build_kernel.sh task executes without any error, but the build_system.sh fails while it's trying to compile QT.
It loads the **ubuntu-base-24.04.1-base-arm64.tar.gz,** which then fails to fetch the necessary apt packages:
`~/projects/agnos-builder$ ./build_system.sh docker images
Registering qemu-user-static
[+] Building 0.5s (3/3) FINISHED                                                                                                                                          docker:default
 => [internal] load build definition from Dockerfile.agnos                                                                                                                          0.0s
 => => transferring dockerfile: 10.25kB                                                                                                                                             0.0s
 => [internal] load metadata for docker.io/library/ubuntu:24.04                                                                                                                     0.5s
 => [internal] load .dockerignore                                                                                                                                                   0.0s
 => => transferring context: 127B                                                                                                                                                   0.0s
Check complete, no warnings found.
Building agnos-builder docker image
[+] Building 5.5s (24/104)                                                                                                                                                docker:default
 => [internal] load build definition from Dockerfile.agnos                                                                                                                          0.0s
 => => transferring dockerfile: 10.25kB                                                                                                                                             0.0s
 => [internal] load metadata for docker.io/library/ubuntu:24.04                                                                                                                     0.1s
 => [internal] load .dockerignore                                                                                                                                                   0.0s
 => => transferring context: 127B                                                                                                                                                   0.0s
 => [internal] load build context                                                                                                                                                   0.8s
 => => transferring context: 154.64MB                                                                                                                                               0.8s
 => [agnos-compiler 1/2] FROM docker.io/library/ubuntu:24.04@sha256:8a37d68f4f73ebf3d4efafbcf66379bf3728902a8038616808f04e34a9ab63ee                                                0.0s
 => CACHED [agnos-compiler 2/2] RUN apt-get update && apt-get install -yq --no-install-recommends     build-essential     ca-certificates     ccache     clang     curl     checki  0.0s
 => CACHED [agnos-compiler-qtwayland5 1/4] COPY ./userspace/qtwayland/*.deb /tmp/agnos/                                                                                             0.0s
 => CACHED [agnos-compiler-pyqt5 2/5] COPY ./userspace/openpilot_python_dependencies.sh /tmp/agnos/                                                                                 0.0s
 => CACHED [agnos-compiler-capnp 1/2] COPY ./userspace/compile-capnp.sh /tmp/agnos/                                                                                                 0.0s
 => CACHED [agnos-compiler-modemmanager 1/2] COPY ./userspace/compile-modemmanager.sh /tmp/agnos/                                                                                   0.0s
 => CANCELED [agnos-compiler-pyqt5 3/5] RUN /tmp/agnos/openpilot_python_dependencies.sh                                                                                             4.3s
 => CACHED [agnos-compiler-ffmpeg 1/2] COPY ./userspace/compile-ffmpeg.sh /tmp/agnos/                                                                                               0.0s
 => CACHED [agnos-base  1/12] ADD ubuntu-base-24.04.1-base-arm64.tar.gz /                                                                                                           0.0s
 => CACHED [agnos-base  2/12] RUN mkdir -p /tmp/agnos                                                                                                                               0.0s
 => CACHED [agnos-base  3/12] RUN set -xe                                                                                                                                           0.0s
 => CACHED [agnos-base  4/12] RUN echo "resolvconf resolvconf/linkify-resolvconf boolean false" | debconf-set-selections                                                            0.0s
 => CACHED [agnos-base  5/12] COPY ./userspace/base_setup.sh /tmp/agnos                                                                                                             0.0s
 => CANCELED [agnos-compiler-capnp 2/2] RUN --mount=type=cache,target=/root/.ccache,id=capnp,sharing=shared     /tmp/agnos/compile-capnp.sh                                         4.3s
 => CANCELED [agnos-compiler-modemmanager 2/2] RUN --mount=type=cache,target=/root/.ccache,id=modemmanager,sharing=shared     /tmp/agnos/compile-modemmanager.sh                    4.2s
 => CANCELED [agnos-base  6/12] RUN /tmp/agnos/base_setup.sh                                                                                                                        4.2s
 => CANCELED [agnos-compiler-ffmpeg 2/2] RUN --mount=type=cache,target=/root/.ccache,id=ffmpeg,sharing=shared     /tmp/agnos/compile-ffmpeg.sh                                      4.4s
 => CACHED [agnos-compiler-qtwayland5 2/4] COPY ./userspace/compile-qtwayland5.sh /tmp/agnos/                                                                                       0.0s
 => CACHED [agnos-compiler-qtwayland5 3/4] COPY ./userspace/qtwayland/patch* /tmp/agnos/                                                                                            0.0s
 => ERROR [agnos-compiler-qtwayland5 4/4] RUN --mount=type=cache,target=/root/.ccache,id=qtwayland5,sharing=shared     /tmp/agnos/compile-qtwayland5.sh                             4.0s
------                                                                                                                                                                                   
 > [agnos-compiler-qtwayland5 4/4] RUN --mount=type=cache,target=/root/.ccache,id=qtwayland5,sharing=shared     /tmp/agnos/compile-qtwayland5.sh:                                        
1.024 Hit:1 http://security.ubuntu.com/ubuntu noble-security InRelease                                                                                                                   
1.083 Hit:2 http://archive.ubuntu.com/ubuntu noble InRelease                                                                                                                             
1.138 Ign:3 http://security.ubuntu.com/ubuntu noble-security/restricted armhf Packages                                                                                                   
1.178 Ign:4 http://security.ubuntu.com/ubuntu noble-security/multiverse armhf Packages                                                                                                   
1.178 Ign:5 http://security.ubuntu.com/ubuntu noble-security/universe armhf Packages                                                                                                     
1.178 Ign:6 http://security.ubuntu.com/ubuntu noble-security/main armhf Packages                                                                                                         
1.198 Hit:7 http://archive.ubuntu.com/ubuntu noble-updates InRelease                                                                                                                     
[... cut ...]
3.804 E: Unable to correct problems, you have held broken packages.
------
Dockerfile.agnos:56
--------------------
  55 |     COPY ./userspace/qtwayland/patch* /tmp/agnos/
  56 | >>> RUN --mount=type=cache,target=/root/.ccache,id=qtwayland5,sharing=shared \
  57 | >>>     /tmp/agnos/compile-qtwayland5.sh
  58 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c /tmp/agnos/compile-qtwayland5.sh" did not complete successfully: exit code: 100
`

Any assistance would be appreciated!

andiradulescu, 2024-09-15:
Thanks for reporting! We are getting this fixed and will report back.

MisterBerg, 2024-09-16:
No problem, I'm happy to help!

closed 2024-09-16 by adeebshihadeh