commaai/agnos-builder #387: `/var/` fills up much faster than before

adeebshihadeh, 2024-10-18:
This is spammed to the journal after 17hrs of uptime:
```
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: file '/var/log/syslog'[8] write error - see https://www.rsyslog.com/solving-rsyslog-write-errors/ for help OS error: No space left on device [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
Oct 18 17:24:37 comma-863276c1 rsyslogd[3590]: action 'action-1-builtin:omfile' (module 'builtin:omfile') message lost, could not be processed. Check for additional error messages before this one. [v8.2312.0 try https://www.rsyslog.com/e/2027 ]
```

adeebshihadeh, 2024-10-24:
Ah
```bash
comma@comma-863276c1:/var$ df -h | grep var
tmpfs           128M  128M     0 100% /var
comma@comma-863276c1:/var$ sudo du -hs /var
5.1M    /var
```

jyoung8607, 2025-01-26:
Seeing this as well on a couple devices here with `DisablePowerDown` set. 

It looks like `logrotate` isn't doing anything at all, not sure why. I think (not sure) this was working on AGNOS 10.

jyoung8607, 2025-01-26:
Confirmed that NetworkManager writes a variety of transient state info in `/var/run` and `/var/lib`, which helps explain a weird network problem I had today. systemd and dbus and udev and a bunch of other services do the same. A full `/var` is going to create all kinds of weird issues.

jyoung8607, 2025-01-26:
Partial correction: `/var/run` is on a separate tmpfs which isn't directly affected and is supported to be on tmpfs. However, NetworkManager and other daemons still store other things in `/var/lib`, like our current DHCP lease. This stuff shouldn't even be on a tmpfs, it's supposed to survive through reboots.

closed 2025-02-04 by adeebshihadeh