commaai/agnos-builder #418: ./build_system.sh failed with "No module named '_cffi_backend'"

zzhuang0816, 2024-12-10:
I tried both on a M1 and M3 mac, and both failed at build_system.sh:
`=> [stage-6 14/74] COPY ./userspace/uv /tmp/agnos/uv                                                                                                                                                                                           0.0s 
 => ERROR [stage-6 15/74] RUN source /usr/local/venv/bin/activate &&     cd /tmp/agnos/uv &&     export PYOPENCL_CL_PRETEND_VERSION="2.0" &&     pc="$(python -c "import sysconfig;print(sysconfig.get_config_vars('installed_base')[0])")" &&  3.4s 
------                                                                                                                                                                                                                                               
 > [stage-6 15/74] RUN source /usr/local/venv/bin/activate &&     cd /tmp/agnos/uv &&     export PYOPENCL_CL_PRETEND_VERSION="2.0" &&     pc="$(python -c "import sysconfig;print(sysconfig.get_config_vars('installed_base')[0])")" &&     pcpath=$pc"/lib/pkgconfig" &&     export PKG_CONFIG_PATH="$pcpath:$PKG_CONFIG_PATH" &&     pypcpath=$pcpath"/python-3.12.pc" &&     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpath &&     pypcpathembed=$pcpath"/python-3.12-embed.pc" &&     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpathembed &&     MAKEFLAGS="-j$(nproc)" UV_NO_CACHE=1 UV_PROJECT_ENVIRONMENT=/usr/local/venv uv sync --inexact --compile-bytecode:                                                                                 
0.263 warning: Missing version constraint (e.g., a lower bound) for `dbus-python`                                                                                                                                                                    
0.263 warning: Missing version constraint (e.g., a lower bound) for `evdev`                                                                                                                                                                          
0.263 warning: Missing version constraint (e.g., a lower bound) for `smbus2`
0.263 warning: Missing version constraint (e.g., a lower bound) for `ipython`
0.263 warning: Missing version constraint (e.g., a lower bound) for `pyrasite`
0.263 warning: Missing version constraint (e.g., a lower bound) for `py-spy`
3.403   × Failed to download and build `aiortc==0.9.27`
3.403   ╰─▶ Build backend failed to determine metadata through
3.403       `prepare_metadata_for_build_wheel` (exit status: 1)
3.403 
3.403       [stderr]
3.403       Traceback (most recent call last):
3.403         File "<string>", line 14, in <module>
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/build_meta.py",
3.403       line 377, in prepare_metadata_for_build_wheel
3.403           self.run_setup()
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/build_meta.py",
3.403       line 522, in run_setup
3.403           super().run_setup(setup_script=setup_script)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/build_meta.py",
3.403       line 320, in run_setup
3.403           exec(code, locals())
3.403         File "<string>", line 35, in <module>
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/__init__.py",
3.403       line 117, in setup
3.403           return distutils.core.setup(**attrs)
3.403                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/_distutils/core.py",
3.403       line 145, in setup
3.403           _setup_distribution = dist = klass(attrs)
3.403                                        ^^^^^^^^^^^^
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/dist.py",
3.403       line 319, in __init__
3.403           _Distribution.__init__(self, dist_attrs)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/_distutils/dist.py",
3.403       line 279, in __init__
3.403           self.finalize_options()
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/dist.py",
3.403       line 677, in finalize_options
3.403           ep(self)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/setuptools/dist.py",
3.403       line 697, in _finalize_setup_keywords
3.403           ep.load()(self, ep.name, value)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/cffi/setuptools_ext.py",
3.403       line 216, in cffi_modules
3.403           add_cffi_module(dist, cffi_module)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/cffi/setuptools_ext.py",
3.403       line 49, in add_cffi_module
3.403           execfile(build_file_name, mod_vars)
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/cffi/setuptools_ext.py",
3.403       line 25, in execfile
3.403           exec(code, glob, glob)
3.403         File "src/_cffi_src/build_opus.py", line 3, in <module>
3.403           ffibuilder = FFI()
3.403                        ^^^^^
3.403         File
3.403       "/tmp/.tmpiPsyuT/builds-v0/.tmpZlBVW7/lib/python3.12/site-packages/cffi/api.py",
3.403       line 48, in __init__
3.403           import _cffi_backend as backend
3.403       ModuleNotFoundError: No module named '_cffi_backend'
3.403 
3.403   help: `aiortc` (v0.9.27) was included because `agnos` (v0.1.0) depends on
3.403         `openpilot` (v0.1.0) which depends on `aiortc`
------
Dockerfile.agnos:133
--------------------
 132 |     COPY ./userspace/uv /tmp/agnos/uv
 133 | >>> RUN source $XDG_DATA_HOME/venv/bin/activate && \
 134 | >>>     cd /tmp/agnos/uv && \
 135 | >>>     export PYOPENCL_CL_PRETEND_VERSION="2.0" && \
 136 | >>>     pc="$(python -c "import sysconfig;print(sysconfig.get_config_vars('installed_base')[0])")" && \
 137 | >>>     pcpath=$pc"/lib/pkgconfig" && \
 138 | >>>     export PKG_CONFIG_PATH="$pcpath:$PKG_CONFIG_PATH" && \
 139 | >>>     pypcpath=$pcpath"/python-3.12.pc" && \
 140 | >>>     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpath && \
 141 | >>>     pypcpathembed=$pcpath"/python-3.12-embed.pc" && \
 142 | >>>     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpathembed && \
 143 | >>>     MAKEFLAGS="-j$(nproc)" UV_NO_CACHE=1 UV_PROJECT_ENVIRONMENT=$XDG_DATA_HOME/venv uv sync --inexact --compile-bytecode
 144 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c source $XDG_DATA_HOME/venv/bin/activate &&     cd /tmp/agnos/uv &&     export PYOPENCL_CL_PRETEND_VERSION=\"2.0\" &&     pc=\"$(python -c \"import sysconfig;print(sysconfig.get_config_vars('installed_base')[0])\")\" &&     pcpath=$pc\"/lib/pkgconfig\" &&     export PKG_CONFIG_PATH=\"$pcpath:$PKG_CONFIG_PATH\" &&     pypcpath=$pcpath\"/python-3.12.pc\" &&     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpath &&     pypcpathembed=$pcpath\"/python-3.12-embed.pc\" &&     sed -i 's|prefix=/install|prefix='$pc'|' $pypcpathembed &&     MAKEFLAGS=\"-j$(nproc)\" UV_NO_CACHE=1 UV_PROJECT_ENVIRONMENT=$XDG_DATA_HOME/venv uv sync --inexact --compile-bytecode" did not complete successfully: exit code: 1
`

BBBmau, 2024-12-10:
This is addressed by reverting the python packages from this PR: https://github.com/commaai/agnos-builder/pull/415

though I wonder if there's something that needs to be done on our side in order to build successfully, the image itself is built just fine when flashing through flash.comma.ai

zzhuang0816, 2024-12-10:
It built successfully now after reverting the python packages.

closed 2024-12-10 by zzhuang0816