commaai/agnos-builder #420: run ./build_system.sh run failed

kan-sir, 2024-12-17:
156.4 Setting up openssl (3.0.13-0ubuntu3.4) ...
156.4 Setting up publicsuffix (20231001.0357-0.1) ...
156.4 Setting up libldap2:arm64 (2.6.7+dfsg-1~exp1ubuntu8.1) ...
156.4 Setting up wget (1.21.4-1ubuntu4.1) ...
156.4 Setting up ca-certificates (20240203) ...
159.0 debconf: unable to initialize frontend: Dialog
159.0 debconf: (TERM is not set, so the dialog frontend is not usable.)
159.0 debconf: falling back to frontend: Readline
159.0 debconf: unable to initialize frontend: Readline
159.0 debconf: (Can't locate Term/ReadLine.pm in @INC (you may need to install the Term::ReadLine module) (@INC entries checked: /etc/perl /usr/local/lib/aarch64-linux-gnu/perl/5.38.2 /usr/local/share/perl/5.38.2 /usr/lib/aarch64-linux-gnu/perl5/5.38 /usr/share/perl5 /usr/lib/aarch64-linux-gnu/perl-base /usr/lib/aarch64-linux-gnu/perl/5.38 /usr/share/perl/5.38 /usr/local/lib/site_perl) at /usr/share/perl5/Debconf/FrontEnd/Readline.pm line 8.)
159.0 debconf: falling back to frontend: Teletype
167.3 Updating certificates in /etc/ssl/certs...
185.0 146 added, 0 removed; done.
185.2 Setting up libgssapi-krb5-2:arm64 (1.20.1-6ubuntu2.2) ...
185.3 Setting up libssh-4:arm64 (0.10.6-2build2) ...
185.3 Setting up libcurl4t64:arm64 (8.5.0-2ubuntu10.6) ...
185.3 Setting up curl (8.5.0-2ubuntu10.6) ...
185.3 Processing triggers for libc-bin (2.39-0ubuntu8.3) ...
185.4 Processing triggers for ca-certificates (20240203) ...
185.5 Updating certificates in /etc/ssl/certs...
195.5 0 added, 0 removed; done.
195.5 Running hooks in /etc/ca-certificates/update.d...
195.6 done.
326.3 /tmp/agnos/base_setup.sh: line 21: apt-fast: command not found
------
Dockerfile.agnos:70
--------------------
  68 |     RUN echo "resolvconf resolvconf/linkify-resolvconf boolean false" | debconf-set-selections
  69 |     COPY ./userspace/base_setup.sh /tmp/agnos
  70 | >>> RUN /tmp/agnos/base_setup.sh
  71 |
  72 |     # Install openpilot dependencies
--------------------
ERROR: failed to solve: process "/bin/sh -c /tmp/agnos/base_setup.sh" did not complete successfully: exit code: 127

robin-reckmann, 2024-12-17:
Does this step reliably fail for you? Could you provide the full logs of the `base_setup.sh` step? Seems like an issue with the installation of `apt-fast`.

kan-sir, 2024-12-17:
> Does this step reliably fail for you? Could you provide the full logs of the `base_setup.sh` step? Seems like an issue with the installation of `apt-fast`.

thank you for your reply！ the issue caused by apt-fast was uninstalled ，please close the issue.

closed 2024-12-17 by kan-sir