commaai/agnos-builder #423: Wi-FI network configuration lost on AGNOS downgrade

jyoung8607, 2025-01-12:
Wi-Fi network configuration is lost when downgrading openpilot from master-ci (AGNOS 11.4) to release (AGNOS 10.1).

**Steps to reproduce**

1) Start with current master-ci on AGNOS 11.4, on a device without comma prime/LTE connectivity
2) Uninstall from the UI
3) Attempt a fresh install of release openpilot (no custom software), as of this writing openpilot 0.9.7
4) openpilot startup spins in "registering device".

**Additional info**

This seems to trace back to a core Ubuntu change for how networks are configured, in NetworkManager vs netplan/systemd. A Wi-Fi network created on AGNOS 10.1 is saved by NetworkManager, but a network created on AGNOS 11.4 is saved by netplan.  

During uninstall and reinstall, /data is wiped and a new Wi-FI network config is collected by the AGNOS version you're coming FROM. Then the requested version of openpilot is fetched, and the matching AGNOS is installed.

NetworkManager in AGNOS 11.4 understands both config styles, so the forward path works, but NetworkManager in AGNOS 10.1 doesn't grok netplan, so the backward path doesn't work. Lack of network connectivity is disruptive to openpilot's first-time startup and registration process.

master-ci, AGNOS 11.4:
```
comma@comma-2a02b9c8:~$ nmcli -f NAME,DEVICE,FILENAME conn show
NAME                        DEVICE    FILENAME
openpilot connection young  wlan0     /run/NetworkManager/system-connections/netplan-NM-9a0a0b65-8745-424e-8475-aed8342d6b9b-young.nmconnection
esim                        cdc-wdm0  /etc/NetworkManager/system-connections/esim.nmconnection
lo                          lo        /run/NetworkManager/system-connections/lo.nmconnection
Hotspot                     --        /run/NetworkManager/system-connections/netplan-NM-d3bf6523-e55b-4838-9175-9a7f6e9e3715-weedle-f39c.nmconnection
lte                         --        /run/NetworkManager/system-connections/lte.nmconnection

comma@comma-2a02b9c8:/data/etc/NetworkManager/system-connections$ ls -al
total 12
drwxr-xr-x 2 root root 4096 Jan 12 14:31 .
drwxr-xr-x 3 root root 4096 Nov 21  2023 ..
-rw------- 1 root root  405 Jan 12 14:24 esim.nmconnection

comma@comma-2a02b9c8:/data/etc/netplan$ ls -al
total 16
drwxr-xr-x 2 root root 4096 Jan 12 14:31 .
drwxr-xr-x 5 root root 4096 Aug  8 14:51 ..
-rw------- 1 root root  741 Jan 12 14:31 90-NM-9a0a0b65-8745-424e-8475-aed8342d6b9b.yaml
-rw------- 1 root root  912 Jan 12 14:24 90-NM-d3bf6523-e55b-4838-9175-9a7f6e9e3715.yaml
```

release3, AGNOS 10.1 after downgrade. My Wi-Fi config is in that YAML file, but NetworkManager doesn't pick it up. Result is the device spinning in "registering device" on startup.

```
comma@comma-2a02b9c8:~$ nmcli -f NAME,DEVICE,FILENAME conn show
NAME  DEVICE    FILENAME                                                    
lte   cdc-wdm0  /usr/lib/NetworkManager/system-connections/lte.nmconnection 

comma@comma-2a02b9c8:/data/etc/NetworkManager/system-connections$ ls -al
ls -al
total 8
drwxr-xr-x 2 root root 4096 Aug  8  2024 .
drwxr-xr-x 3 root root 4096 Aug  8  2024 ..

comma@comma-2a02b9c8:/data/etc/netplan$ ls -al
ls -al
total 12
drwxr-xr-x 2 root root 4096 Aug  8  2024 .
drwxr-xr-x 5 root root 4096 Aug  8  2024 ..
-rw------- 1 root root  741 Aug  8  2024 90-NM-dc689106-c80b-4741-a926-e7e0918d709a.yaml
```

release3, AGNOS 10.1 after a factory reset allowing me to configure Wi-Fi again.

```
comma@comma-2a02b9c8:~$ nmcli -f NAME,DEVICE,FILENAME conn show
nmcli -f NAME,DEVICE,FILENAME conn show
NAME                        DEVICE    FILENAME                                                                       
lte                         cdc-wdm0  /run/NetworkManager/system-connections/lte.nmconnection                        
openpilot connection young  wlan0     /etc/NetworkManager/system-connections/openpilot connection young.nmconnection 
esim                        --        /etc/NetworkManager/system-connections/esim.nmconnection 

comma@comma-2a02b9c8:/data/etc/NetworkManager/system-connections$ ls -al
ls -al
total 16
drwxr-xr-x 2 root root 4096 Jan 12 15:33  .
drwxr-xr-x 3 root root 4096 Nov 21  2023  ..
-rw------- 1 root root  405 Jan 12 15:33  esim.nmconnection
-rw------- 1 root root  376 Nov 21  2023 'openpilot connection young.nmconnection'

comma@comma-2a02b9c8:~$ ls -al /data/etc/netplan
ls -al /data/etc/netplan
ls: cannot access '/data/etc/netplan': No such file or directory
```

jyoung8607, 2025-01-12:
The correct move here seems to be reconfiguring AGNOS 11 to keep using NetworkManager for configuration. That will eliminate the backward compatibility issue and sidestep the registration hang.

Extremely brief testing only, but this seems to work:

https://www.cyberciti.biz/faq/change-netplan-renderer-from-networkd-to-networkmanager/

[image]

AGNOS will need to stop symlinking /etc/netplan to /data/etc/netplan, and replace it with this static config file.

deanlee, 2025-01-13:
I experienced this issue a few days ago, where the device got stuck on the 'registering device...' screen. I had to connect via serial and resolve it through the serial console.

jyoung8607, 2025-01-15:
I've made some minor changes to the repro conditions and steps to reproduce.

Unfortunately the issue is a little more insidious. Ubuntu 24.04 / Noble Numbat ships with a modified NetworkManager that explicitly tries to use netplan for new config storage. This is why /etc/netplan being readonly caused #407. My proposed config change above was defeating the netplan migration, but would reintroduce #407 and related problems.

* [NetworkManager on Noble Numbat](https://git.launchpad.net/ubuntu/+source/network-manager/log/?h=applied/ubuntu/noble)
  * [netplan patch 1](https://git.launchpad.net/ubuntu/+source/network-manager/commit/?h=applied/ubuntu/noble&id=f9a0431bdb251c4f852f83f2a063da6763d512c6)
  * [netplan patch 2](https://git.launchpad.net/ubuntu/+source/network-manager/commit/?h=applied/ubuntu/noble&id=2096da3a84e74a60af670c475604faa9cd840ffc)

**Options**

* Build NetworkManager ourselves with `enable_netplan=no` (they did at least make it a build option). We do already ship a custom build of ModemManager.
* Add a last-step hook to the AGNOS update flow, to always convert any netplan-only network configs to NetworkManager keyfiles. Both are very simple text files, and we only care about two strings (SSID and PSK).
* Take 500mg of fukitol and close the bug. This issue annoys me at a visceral level, and I think it should be fixed, but the sun will rise tomorrow if a won't-fix decision is reached.

jyoung8607, 2025-01-15:
AGNOS 10 can only read and write NetworkManager keyfiles.
AGNOS 11 can read either format, but will only write netplan YAML.

**Example NetworkManager keyfile**

```
[connection]
id=openpilot connection young
uuid=eb38d9ea-ebb3-4160-a373-0ac2e63fe4ba
type=wifi
autoconnect-retries=0
permissions=

[wifi]
mac-address-blacklist=
mode=infrastructure
ssid=young

[wifi-security]
auth-alg=open
key-mgmt=wpa-psk
psk=[wifi-password-in-cleartext]

[ipv4]
dns-priority=600
dns-search=
method=auto

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=ignore

[proxy]

```

**Example netplan YAML**

```
network:
  version: 2
  wifis:
    NM-d4eabbc2-d9f1-4603-941a-dfd42d2db814:
      renderer: NetworkManager
      match: {}
      dhcp4: true
      access-points:
        "young":
          auth:
            key-management: "psk"
            password: "[wifi-password-in-cleartext]"
          networkmanager:
            uuid: "d4eabbc2-d9f1-4603-941a-dfd42d2db814"
            name: "openpilot connection young"
            passthrough:
              connection.autoconnect-retries: "0"
              wifi-security.auth-alg: "open"
              ipv4.dns-priority: "600"
              ipv6.addr-gen-mode: "default"
              proxy._: ""
      networkmanager:
        uuid: "d4eabbc2-d9f1-4603-941a-dfd42d2db814"
        name: "openpilot connection young"
```

adeebshihadeh, 2025-01-16:
Going with option #3 unless something simple comes along.

closed 2025-01-16 by adeebshihadeh