commaai/agnos-builder #455: Errors while running build_system.sh (ffmpeg, qtwaland5, modemmanager)

chebilyoussef21, 2025-02-08:
Issue:
I'm encountering an issue when building the agnos system, specifically when compiling the ffmpeg, qtwaland packages). For some reason, using the exact same script, the build works completely fine when running the ./build_system.sh file on an Ubuntu 20.04 AMD computer with older packages but fails on an AMD laptop running Ubuntu 24.04. 
The build process will take approximately 1000 seconds before crashing and giving an error similar to the following but the compile script will vary from build to build (compile-ffmpeg once compile-qtwaland5 another time):
Dockerfile.agnos:47
--------------------
  46 |     COPY ./userspace/qtwayland/patch* /tmp/agnos/
  47 | >>> RUN --mount=type=cache,target=/root/.ccache,id=qtwayland5,sharing=shared \
  48 | >>>     /tmp/agnos/compile-qtwayland5.sh
  49 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c /tmp/agnos/compile-qtwayland5.sh" did not complete successfully: exit code: 2

Specifics:
The ffmpeg compilation errors seem to relate to mismatches in the dependencies maybe (example below):  
./libavutil/linm.h. error: static declaration of cbrt follows non static declaration 
As for the qtwayland errors, there is no clear 'logged' indication as to why it is crashing but I do get warnings related to deprecated functions (example below):
690.8 /tmp/qtwayland/src/client/qwaylanddatadevice.cpp:197:43: warning: 'static QPlatformDragQtResponse QWindowSystemInterface::handleDrag(QWindow*, const QMimeData*, const QPoint&, Qt::DropActions)' is deprecated [-Wdeprecated-declarations]
However, even after removing warning detection in the compile-qtwayland5.sh script, by setting CXXFLAGS="-Wno-deprecated-declarations" within the script, it still crashes during that step of the build process.

Package Versions:
In terms of package versions, the 20.04 computer had gcc 9 and ffmpeg version 4.2.2 and the 24.04 computer has gcc13 with ffmpeg 7.1 (had version 6.1.1 before I updated it to 7.1 to see if that would resolve the problem). I've tried updating all the libraries but that doesn't seem to solve anything.

Opinion:
The build system relies on Docker so I don't see why version mismatches would be a problem for this. Also, it would make more sense if the outdated packages in Ubuntu 20.04 were causing the problems but not the other way around. I've checked the Discord channels and the other issues posted here, but there seems to be no other occurrence of this issue despite many stating they built agnos on their Ubuntu 24.04 devices. 

If anyone has any idea on how to fix this or what the cause could be, please let me know. I'd really appreciate some guidance here. Thanks.

adeebshihadeh, 2025-02-08:
The AGNOS build now requires building on a native ARM64 machine.

chebilyoussef21, 2025-02-08:
> The AGNOS build now requires building on a native ARM64 machine. We use namespace.so.
@adeebshihadeh 
Interesting. But why did it work on the Ubuntu 20.04 amd64 computer then, with the exact same scripts and steps?

robin-reckmann, 2025-04-04:
@chebilyoussef21 Only difference between your builds on the two machines should be the docker version you are running. The whole agnos build is done in the docker, so the environment should not differ between the two machines. Could you provide more info on the docker versions? While its still technically possible to compile on a non arm64 architecture, its very slow.

robbiewu008, 2025-04-26:
I encounter the problem of segfault  on ubuntu 22.04 x86_64 system. it is OK on MacOS arm.
------------------------------------------------
[771564.119125] docker0: port 4(veth3ee4baa) entered blocking state
[771564.119128] docker0: port 4(veth3ee4baa) entered forwarding state
[771947.954796] python3.12[3307830]: segfault at 371f350 ip 000000000061c4a0 sp 00007ffe4dcccde8 error 4 in qemu-aarch64-static[401000+462000]
[771947.954808] Code: 06 09 18 00 66 0f 1f 44 00 00 64 83 2c 25 50 ff ff ff 01 74 05 c3 0f 1f 40 00 bf e0 86 e0 00 e9 56 16 18 00 66 0f 1f 44 00 00 <64> 8b 04 25 50 ff ff ff 85 c0 0f 9f c0 c3 66 90 48 83 ec 08 64 8b
[771949.244792] docker0: port 2(veth48946ac) entered disabled state
[771949.244875] veth2c3737f: renamed from eth0

robbiewu008, 2025-04-26:
segfault can happen randomly with any application in qemu-aarch64-static, like gcc, g++, python etc.
@chebilyoussef21 actually it is not the scripts issue, the qemu issue, have a look at your dmesg log to check the segfault.

adeebshihadeh, 2025-10-04:
> [@adeebshihadeh](https://github.com/adeebshihadeh) Interesting. But why did it work on the Ubuntu 20.04 amd64 computer then, with the exact same scripts and steps?

Not sure since we always build with native. We'd accept a PR to fix the issue, but we don't plan on fixing this ourselves so I'm going to close.

closed 2025-10-04 by adeebshihadeh