commaai/agnos-builder #464: uv python is slow at decompressing xz updates

andiradulescu, 2025-03-12:
Update is still terribly slow.

https://github.com/commaai/openpilot/pull/33511
https://github.com/astral-sh/python-build-standalone/issues/327

jyoung8607, 2025-03-12:
Are you sure this is a current problem? That history is from September, and due to a *different* uv python bug, AGNOS went to system Python in January. 

https://github.com/commaai/agnos-builder/pull/431
https://github.com/commaai/openpilot/issues/33993

andiradulescu, 2025-03-12:
Thanks, I now remember that problem. That’s really good news, since `uv python` is problematic. Leaving this open until I check why it still seems slow to me.

andiradulescu, 2025-03-12:
Weirldy deleter seemed inconsistent recently, too - deleting newer segments instead of older ones. Again, I might be wrong. Don’t have much time to investigate, but will try.

andiradulescu, 2025-03-12:
I renamed the issue, I flashed the exact system image from AGNOS 10.1 and from AGNOS 11.10 via the startup installer:
- 9min on AGNOS 11
- 3min45 on AGNOS 10

royjr, 2025-03-17:
Definitely slow. Uninstalled from menu, reset failed and rebooted (probably unrelated). The wifi selection screen takes forever to populate, after entering the password, takes forever to connect, sometimes fails to connect unless tapped on again. Installing stuck on 0% for at least 2 minutes. Brought it inside and conencted using ethernet adapter and much better.