commaai/agnos-builder #467: Unable to flash Comma 3 with Ubuntu

gilbreen, 2025-04-08:
I am attempting to flash a Comma 3 using under Ubuntu using the steps listed on the repository page.  The following commands issue without error:

```
git clone https://github.com/commaai/agnos-builder.git
cd agnos-builder
git submodule update --init agnos-kernel-sdm845
./tools/extract_tools.sh
```

I was able to build the kernel successfully, but I received the following error when issuing the command to build the system:

```
1423.3 E: Sub-process /usr/bin/dpkg returned an error code (1)

Dockerfile.agnos:33

  32 |     COPY ./userspace/compile-ffmpeg.sh /tmp/agnos/
  33 | >>> RUN --mount=type=cache,target=/root/.ccache,id=ffmpeg,sharing=shared \
  34 | >>>     /tmp/agnos/compile-ffmpeg.sh
  35 |     

ERROR: failed to solve: process "/bin/sh -c /tmp/agnos/compile-ffmpeg.sh" did not complete successfully: exit code: 100
```

Since flashing from building wasn't working, I followed the recommendation on the github page to issue the command:

`sudo ./download-from-manifest.py`

and was able to successfully download the image files.

I then connected the Comma 3 in the order instructed and issued the following command:

`for d in /sys/bus/usb/drivers/qcserial/*-*; do [ -e "$d" ] && echo -n "$(basename $d)" | sudo tee /sys/bus/usb/drivers/qcserial/unbind > /dev/null; done`

I then issued the command:

`./flash_all.sh`

The process took while and progresses to 100% but then had the following output:

```
Wrote /home/gilbert/agnos-builder/output/system.img to sector 8712.
Qualcomm Sahara / Firehose Client V3.62 (c) B.Kerler 2018-2023.
main - Trying with no loader given ...
main - Waiting for the device
main - Device detected :)
main - Mode detected: firehose
INFO: Will issue reset/power off 100 useconds, if this hangs check if watchdog is enabled
INFO: bsp_target_reset() 0
DeviceClass - USBError(19, 'No such device (it may have been disconnected)')
Flashed system_a!
```

The Comma 3 then reboots and again shows the Comma logo with the message "Press any to shutdown"

I am not sure if I am missing steps or need to do something different.  Any guidance is appreciated.

stejpet, 2025-07-10:
I can confirm this. Exactly the same result.

gilbreen, 2025-07-10:
Thanks.  The comma team fixed the issue a few months ago and I was able to successfully flash my device using the flash.comma.ai website.  It was fixed under the flash repository.

https://github.com/commaai/flash/issues/132